riscv-non-isa/riscv-rvv-intrinsic-doc #158: Question about masked vredsum

Mousavikia, 2022-08-06:
Hi guys... 
I have a vector like this:
`signed short int in[16] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16 };`
I want to reduction some this to 4 numbers:
Out[0] = 1+2+3+4
Out[1] = 5+6+7+8
Out[2] = 9+10+11+12
Out[3] = 13+14+15+16
I want to utilize masked `vredsum `although I know I can use `vslide` as well...
I have written the following code but after executing it on a sample vector coprocessor the answer for all out[0] to out[3] is 10 which is 1+2+3+4 meaning it just adds 4 beginning numbers of the vector discarding `vl` and mask!
The code:
```
#include <uart.h>
#include <runtime.h>
#include <riscv_vector.h>
#include <stddef.h>
void* memcpy(void* dest, const void* src, size_t n)
{
    for (size_t i = 0; i < n; i++)
    {
        ((char*)dest)[i] = ((char*)src)[i];
    }
}
void redsum(signed short int* in, signed short int* out) {
    uint16_t mask1 = 0b1111000000000000;
    uint16_t mask2 = 0b0000111100000000;
    uint16_t mask3 = 0b0000000011110000;
    uint16_t mask4 = 0b0000000000001111;
    vint16m1_t vin = vle16_v_i16m1(in, 16);

    vint16m1_t vec_zero1 = vmv_v_x_i16m1(0, 16);
    vbool16_t vmask1 = vlm_v_b16((uint8_t *)&mask1, 16);
    vint16m1_t vec_sum1 = vredsum_vs_i16m1_i16m1_m(vmask1, vec_zero1, vin, vec_zero1, 16);
    out[0] = vmv_x_s_i16m1_i16(vec_sum1);

    vint16m1_t vec_zero2 = vmv_v_x_i16m1(0, 16);
    vbool16_t vmask2 = vlm_v_b16((uint8_t *)&mask2, 16);
    vint16m1_t vec_sum2 = vredsum_vs_i16m1_i16m1_m(vmask2, vec_zero2, vin, vec_zero2, 16);
    out[1] = vmv_x_s_i16m1_i16(vec_sum2);

    vint16m1_t vec_zero3 = vmv_v_x_i16m1(0, 16);
    vbool16_t vmask3 = vlm_v_b16((uint8_t *)&mask3, 16);
    vint16m1_t vec_sum3 = vredsum_vs_i16m1_i16m1_m(vmask3, vec_zero3, vin, vec_zero3, 16);
    out[2] = vmv_x_s_i16m1_i16(vec_sum3);

    vint16m1_t vec_zero4 = vmv_v_x_i16m1(0, 16);
    vbool16_t vmask4 = vlm_v_b16((uint8_t *)&mask4, 16);
    vint16m1_t vec_sum4 = vredsum_vs_i16m1_i16m1_m(vmask4, vec_zero4, vin, vec_zero4, 16);
    out[3] = vmv_x_s_i16m1_i16(vec_sum4);
}
int main() {
    signed short int out[4];
    signed short int in[16] = { 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16 };
    redsum(in, out);
    uart_printf("Out1 is: %d\n\r", out[0]);
    uart_printf("Out2 is: %d\n\r", out[1]);
    uart_printf("Out3 is: %d\n\r", out[2]);
    uart_printf("Out4 is: %d\n\r", out[3]);
    asm volatile("jr x0;");
    return 0;
}
```
Can someone help me?

eopXD, 2022-08-09:
Your casting to `uint16_t` to `uint8_t*` is not correct.
Please note that each element of the mask register will be loaded from an element of the `uint8_t*` address.

closed 2022-08-09 by eopXD

nick-knight, 2022-08-09:
I ran this code (slight modification for `printf`) on SiFive's X280 QEMU model, using SiFive's LLVM toolchain, and got the following output:
```
Out1 is: 58
Out2 is: 42
Out3 is: 26
Out4 is: 10
```
Appears to be correct.

The only issue I see in this code is that it assumes VLEN >= 16*16 = 256, whereas the V extension only requires VLEN >= 128.

topperc, 2022-08-09:
It looks like mask1 and mask4 need to be swapped and mask2 and mask3 need to be swapped. `0b` prefixed bit strings have MSB on the left and LSB on the right.

Mousavikia, 2022-08-09:
Hi @eopXD , @topperc , @nick-knight 
Thank you all for your comments...
@topperc  Yeah I confused the order of the masks... Thank you.
@nick-knight For the issue you mentioned `The only issue I see in this code is that it assumes VLEN >= 16*16 = 256, whereas the V extension only requires VLEN >= 128.` I think I should use `lmul `since I have a small register file (mine is 64 since I have Zve32x minimum VLEN is 32 in width so `lmul=4` I think is ok)
@eopXD About the casting can you say what should I write instead? Should I delete the casting?
`vbool16_t vmask1 = vlm_v_b16(mask1, 16);` Sorry I cannot understand the second line of your comment

topperc, 2022-08-09:
> Hi @eopXD , @topperc , @nick-knight Thank you all for your comments... @topperc Yeah I confused the order of the masks... Thank you. @nick-knight For the issue you mentioned `The only issue I see in [...]

Your use of `vlm_v_b16` was correct. It reads bits of a mask not bytes.

Mousavikia, 2022-08-10:
Thank you @topperc.