riscv-non-isa/riscv-rvv-intrinsic-doc #159: encode return type if necessary

HanKuanChen, 2022-09-04:
We encode return type in intrinsics name (e.g., `vadd_vv_i8m1`) as a general rule.
If intrinsics may have same name, we encode input type together (e.g., `vreinterpret_v_i8mf8_u8mf8`).
But for the following intrinsics, encode input type is enough to distinguish the intrinsic. Can we remove return type? It is superfluous.

```
vmadc_vv_i8m1_b8 -> vmadc_vv_i8m1
vmseq_vv_i8m1_b8 -> vmseq_vv_i8m1
vredsum_vs_i8m1_i8m1 -> vredsum_vs_i8m1
vmv_x_s_i8m1_i8 -> vmv_x_s_i8m1
```

reference: https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/master/rvv-intrinsic-rfc.md#exceptions-in-naming

lhtin, 2022-09-06:
Maybe it is easier to understand that let all intrinsics support the same general name rule (i.e. encoding return type in the last part of the intrinsic name) instead of sometimes encoding return type and sometimes encoding input type in the last part of the name.

HanKuanChen, 2022-09-06:
Yeah, I think it is the reason. However, it is still weird.
What makes `vmv.x.s` to `vmv_x_s_i8mf8_i8` instead of `vmv_x_s_i8mf8` but makes `vcpop.m` to `vcpop_m_b1` instead of `vcpop_m_b1_ulong` or `vcpop_m_b1_unsigned_long` ?

lhtin, 2022-09-07:
Indeed, Maybe we can consider adding the return type to the intrinsics name of `vcpop.m` and `vfirst.m`. Then in the end only store and tuple load do not contain the return type, because their return type is void.

HanKuanChen, 2022-09-07:
Or, we make `vmv_x_s_i8mf8_i8` to `vmv_x_s_i8mf8`. Encode return type only when the type is a vector type.

eopXD, 2023-10-24:
I think it would be a bit rough to introduce a incompatible change now. I think we will settle to what we have now in the specification.

closed 2023-10-24 by eopXD