riscv-non-isa/riscv-rvv-intrinsic-doc #160: mask and maskedoff order for segment load

HanKuanChen, 2022-09-06:
Current design add `mask` and `maskedoff` in the begining of parameter.
```
vint8mf8_t vadd_vv_i8mf8 (vint8mf8_t op1, vint8mf8_t op2, size_t vl);
vint8mf8_t vadd_vv_i8mf8_m (vbool64_t mask, vint8mf8_t maskedoff, vint8mf8_t op1, vint8mf8_t op2, size_t vl);
```
But segment load does not follow the rule
```
void vlseg2e8_v_i8mf8 (vint8mf8_t *v0, vint8mf8_t *v1, const int8_t *base, size_t vl);
void vlseg2e8_v_i8mf8_m (vint8mf8_t *v0, vint8mf8_t *v1, vbool64_t mask, vint8mf8_t maskedoff0, vint8mf8_t maskedoff1, const int8_t *base, size_t vl);
```

Should it be this form?
```
void vlseg2e8_v_i8mf8_m (vbool64_t mask, vint8mf8_t maskedoff0, vint8mf8_t maskedoff1, vint8mf8_t *v0, vint8mf8_t *v1, const int8_t *base, size_t vl);
```

lhtin, 2022-09-06:
I think the current version is better because `v0` and `v1` are not parameter actual, but are used to receive the output of the intrinsic.

HanKuanChen, 2022-09-22:
OK. You convince me.

eopXD, 2022-09-22:
> I think the current version is better because v0 and v1 are not parameter actual, but are used to receive the output of the intrinsic.

@lhtin What you imply is that we put arguments that should receive the output in the arguments first. This does not match interface of fault-only-first-load intrinsics.

```
intrinsic_funcs.md:vuint8mf8_t vle8ff_v_u8mf8 (const uint8_t *base, size_t *new_vl, size_t vl);
```

------
Maybe we want to unify logics behind argument ordering, which may result in modification to the current interface. However since we are going for v1.0 release I think raising this discussion is good.

lhtin, 2022-09-23:
> This does not match interface of fault-only-first-load intrinsic.

Indeed, these fault-only-first-load intrinsics have two return values and are different from `vlseg2e8_v_i8mf8`. `vlseg2e8_v_i8mf8` is a compromised version of the tuple type version `vlseg2e8_v_i8mf8x2` which only have one return value. Therefore, I think these fault-only-first-load intrinsics can be treated separately, not necessarily unified with these which only have one return value. Maybe moving the `*new_vl` argument to the last position is better? I'm not very sure.

eopXD, 2023-10-24:
I think we have settled to what we have now.

closed 2023-10-24 by eopXD