riscv-non-isa/riscv-rvv-intrinsic-doc #162: General rules to follow for actions on an "Issue" or a "PR"

eopXD, 2022-09-08:
As contributors vary time to time, we may want to explicitly list the developing conventions out so we can make sure current or future items are handled in an appropriate manner.

Approximately Issue or PR can fall into these categories:
- Add new API
- Remove existing API
- Change of API semantic
- Document improvement

To any issue concerning adding/removing /changing semantics of the API-s, it is required that the change should be backed with a Proof-Of-Concept in either GCC or LLVM. It goes without saying that discussions are not limited to this but at least a POC is needed before we consider to apply any change.

To make sure implementation is both possible and reasonable, an implementation related change is required to seek at least one approval from an active GCC and at least one approval from an active LLVM developer. An active developer is someone we recognize that contributes to maintenance and is familiar with RISC-V of GCC or LLVM. 

To any issue concerning modification of the documents, the change is required to seek for at least one approval from an active document maintainer.

Apart from the above criteria, it is expected that all Issue / PR approaches to resolve the conflicts and concerns raised. When discussion comes to a deadlock, the active coordinator of this document shall attempt to clarify existing objections, coordinate between discussion participants and push the progress forward to either approving or rejecting the proposal.

------

I think once we reach some consensus on this, we should have the rules explicitly written on some place in this repository. We can discuss offline through this threads and have a final confirmation in the monthly open meeting.

Note: This issue only tries to discuss on how a change should be approved, the detail of applying the change, specifically a backward incompatible change should be further debated in a separate thread, specifically in https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/145.

eopXD, 2022-09-08:
I can't find every Github account of the participants, tagging those that I can currently find and I will attempt to gather everyone's contact in the next open meeting.

Please don't hesitate to share your thoughts.

@rofirrim @frasercrmck @lhtin @kito-cheng @arcbbb

zhongjuzhe, 2022-09-09:
I hope no matter what changes you want to make (including adding new api, change the existing api, ...,etc). Plz update the document first (Don't wait until LLVM implementation) and release V1.0 before November this year. I am recently working on pushing codes to GCC upstream and GCC13 is frozen in November. Thank you!

eopXD, 2022-09-11:
Thank you for the response.

I agree with you that the document update should not wait for full implementation support in the compiler. Just edited my wordings above to emphasize that we just need a POC to proceed on applying an approved proposal.

lhtin, 2022-09-13:
@eopXD Thank you for the proposal for this general rule. Is there miss a category for tools update which include intrinsic generator update?

eopXD, 2022-09-14:
@lhtin Good point. Maintenance of the intrinsic generator is not considered above.

As my colleague @HanKuanChen is drafting the first version of the intrinsic, we should defer this discussion until we have a clear view of what the generator is held responsible for generating. (a list of intrinsics, or even more?)

eopXD, 2022-09-14:
@lhtin Good point. Maintenance of the intrinsic generator is not considered above.

As my colleague @HanKuanChen is drafting the first version of the intrinsic, I think we should defer this discussion until we have a clear view of what the generator is held responsible for generating. (a list of intrinsics, or even more?)

closed 2023-10-24 by eopXD