riscv-non-isa/riscv-rvv-intrinsic-doc #166: Should vcpop.m return type be size_t?

HanKuanChen, 2022-09-15:
Current interface
```
unsigned long vcpop_m_b1 (vbool1_t op1, size_t vl);
```

However, the semantic of vcpop.m is
>  The `vcpop.m` instruction counts the number of mask elements of the active elements of the vector source mask register that have the value 1 and writes the result to a scalar `x` register.

That means the maximum value depends on VL. And VL related interface is
```
size_t vsetvl_e8mf8 (size_t avl);
size_t vsetvlmax_e8mf8 ();
```

So, should we change the interface to
```
size_t vcpop_m_b1 (vbool1_t op1, size_t vl);
```

jan-wassenberg, 2022-09-15:
Sounds good to me.

nick-knight, 2022-09-15:
I've always disagreed with using implementation-defined `size_t` for VL, and would prefer something like https://github.com/riscv-non-isa/riscv-c-api-doc/pull/14 that more closely matches the architecture: VL is defined as an XLEN-bit CSR and its value is always exchanged through (XLEN-bit) X-registers.

My disagreement, however, is very mild: C99 requires `size_t` be at least 16 bits, and of course is XLEN bits in our standard ABIs.

I have a related complaint for vfirst.m. Currently we use `long`, which again is implementation defined but the C language guarantees it has sufficient range for our purposes, and also it is XLEN bits in our standard ABIs.

So I support the proposal here, and support maintaining the status quo for vfirst.m, but still think the XLEN-width types provide a better solution.

PR #287 (2023-10-25): Add more details in the specification
While cleaning up the issue threads, I found some more details we need to have in the specification.

This PR:
- Resolves #74 by adding a description of "Semantic of assignment (`=`) in the C language" under the programming note section of the spec
- Resolves #242 by adding a description of "Mixing inline assembly and intrinsics" under the programming note section of the spec
- Replace `float16_t` with `_Float16` that is now supported by both LLVM and GCC.  This keeps us away from the need to define our own type. (Resolves #31, and alongside also resolves  [... cut]
closed 2023-10-25 by eopXD