riscv-non-isa/riscv-rvv-intrinsic-doc #170: Question on using the intrinsics on mask registers

christian-herber-nxp, 2022-10-05:
Using the current definitions of intrinsics, i am trying to work with some vector masks. From the RVV ISA, I would expect to be able to also use "non-mask" instructions, e.g. to move the values of a mask register to an integer register, compare two masks for equality or whatnot. Is this assumption correct? If so, wouldn't such intrinsics be missing in the current definition of things?

eopXD, 2022-10-05:
You are right, currently we don't have type case from a mask vector register to an integer vector register. Does intrinsics of mask logical instruction suit your need? If not, I am quite curious on the specific use-case you are facing that needs such reinterpretation.

christian-herber-nxp, 2022-10-05:
the conversion to an integer could be useful to integrate RVV into existing SW libraries that e.g. use SSE2. Where you just want to replace the SIMD instructions with vector, but all the traversing through mask results happens with the scalar pipeline. As things are generated, I would think it is best to provide functions anything the hardware is capable of, unless this leads into numbers of permutations that cause issues.

nick-knight, 2022-10-05:
The V extension does not have dedicated mask-to-integer conversion instructions. Indeed, it is impossible to to this generally, because a mask uses (up to) VLEN bits, and VLEN often exceeds XLEN.

If it is known that VL <= M := min(XLEN, ELEN), then an assembly programmer can use an SEW=M vslide1down.vx or vmv.x.s, perhaps with some additional logic to deassert the tail mask bits if necessary. In this manner, multiple slides can be used to unpack the mask register in M-bit chunks into X-registers. Unfortunately, the current intrinsics API currently doesn't support converting (type-punning) vbool*_t to other vector types. Further reading:
https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/61
https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/69
https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/156

christian-herber-nxp, 2022-10-06:
thanks for the detailed elaboration on the mask to integer conversion and point taken.

jan-wassenberg, 2022-11-30:
@nick-knight I have another use case for the other direction: splatting a scalar value into a mask reg.
For example in OddEven, where we want a mask equal to (vid.v & 1) == 0, i.e. 0x55..55.
It would be nice to replace those 3 instructions with a splat + reinterpret.

nick-knight, 2022-11-30:
@jan-wassenberg Thanks for the use-case. For visibility, I cross-posted it to #156, which I think is more directly relevant than this thread.

eopXD, 2023-10-24:
We have the typecast between m1 registers and boolean value now.

closed 2023-10-24 by eopXD