riscv-non-isa/riscv-rvv-intrinsic-doc #175: Confusion about intrinsic naming logic and Vlen setting

Walvka, 2022-11-07:
I am a beginner in rvv intrinsic  and after I started learning I noticed a question about naming.I noticed the naming rules of the current function, eg vadd_e8m1, Is this naming compatible with different chip architectures?  If I change the length of the V register(VLEN), will this function still work? I mean If I write an algorithm, will the program still work when I transform the chip? I think VLEN should be a fixed parameter in the chip, and the corresponding compiler should also fix this parameter, so I tried to search for the VLEN setting method, but I only found some variables in llvm/lib/target/RISCV, and did not find the specific place to determine the length.

Maybe my understanding of RVV is biased, I hope you can help me answer my confusion, thank you very much! Also, this is my first question, if there are any mistakes, I hope you will understand.

eopXD, 2022-11-07:
Minimum VLEN can be specified through the `zve*b` sub-extension. You may refer to v-spec [0] or the LLVM user guide [1] for it.

[0] https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc
[1] https://llvm.org/docs/UserGuides.html

Walvka, 2022-11-07:
> Minimum VLEN can be specified through the `zvl*b` sub-extension. You may refer to v-spec [0] or the LLVM user guide [1] for it.
> 
> [0] https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc [1] https://llvm.org/docs/UserGuides.html

Your advice helped me a lot! But I'm still confused. If I change the sub-extension from Zvl128b to Zvl256b, will the program work correctly after recompiling? For example, vint8m1_t vadd_vv_i8m1(vint8m1_t vs2, vint8m1_t vs1), when the argument is Zvl128b, the addition of SEW==8, LMUL==1 is performed, and I think it is 16 parallel additions with 8 bits per arithmetic unit. When the parameter is Zvl256b, the same addition of SEW==8 and LMUL==1 is performed, but at this time I think it is 32 parallel additions with 8 bits per arithmetic unit. The same function, but the meaning of the expression is different, does this not create compatibility problems?

eopXD, 2022-11-07:
You should code under the a VLA scheme, making sure your code will work upon different VLEN. Feel free to check out the examples [0].

[0] https://github.com/riscv-non-isa/rvv-intrinsic-doc/tree/master/examples

closed 2023-01-11 by eopXD