riscv-non-isa/riscv-rvv-intrinsic-doc #177: Question about vmv.s.x and vmv.s.f's intriniscs with the suffix `m2_tu`, `m4_tu` and `m8_tu`

lhtin, 2022-11-09:
Hi,

From the [rvv spec 16.1 & 16.2 section](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#161-integer-scalar-move-instructions) about `vmv.s.x` and `vmv.s.f`'s description:

> The instructions ignore LMUL and vector register groups.
> ...
> The other elements in the destination vector register ( 0 < index < VLEN/SEW) are treated as tail elements using the current tail agnostic/undisturbed policy.

 I think there should not have `vmv.s.x` and `vmv.s.f`'s intrinsics with the suffix `m2_tu`, `m4_tu` and `m8_tu`, because the tail elements length never exceeds m1 from the rvv spec.

Current intrinsics:

https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/5818330c298306714fefb705a524ddf3b9b50125/policy_funcs/intrinsic_funcs/09_vector_permutation_functions.md?plain=1#L13-L15

Thanks,
Lehua

topperc, 2022-11-10:
If we don't have the intrinsics the user will need to use `vlmul_ext` and `vlmul_trunc` around an m1_tu intrinsic.

I believe this will also cause a problem for the compiler. The compiler uses the intrinsic name to know what SEW and LMUL go with the VL that is passed to it. If you use a VL that came from a vsetvl with LMUL>1, the vl will not be valid for LMUL=1 intrinsic.

nick-knight, 2022-11-10:
I think there is an asymmetry with the reduction instructions, where the current intrinsics only support destination EMUL = 1 variants, and users have to perform LMUL extension/truncation.

> If you use a VL that came from a vsetvl with LMUL>1, the vl will not be valid for LMUL=1 intrinsic.

I agree in general. But in the specific case of `vmv.s.x` and `vfmv.s.f`, all VL > 1 are treated the same by hardware as VL = 1. Whether or not there is a compiler problem, I don't know.

I don't see any value in having destination EMUL != 1 variants of `vmv.s.x` and `vfmv.s.f`, regardless of tail policy.

lhtin, 2022-11-11:
Thank @topperc and @nick-knight for your detailed analysis.

After further thinking, I found that because the specification always treats EMUL=1 of vmv.s.x, the part of the register that exceeds EMUL=1 will retain the original value. That is to say, when I call `vmv_s_x_i8m2_tu(merge, x, vl)`, the following assembly will be generated, assuming:

```asm
vsetvlivli  zero, a1, e8, m2, tu, ma
vmv.s.x v0, a0
```

In this case, since vmv.s.x only modifies the value of the v0 register, the value of v1 is naturally retained. Therefore, it is also in line with the `tu` tail policy. On the contrary, for the intrinsic of `vmv_s_x_i8m2_ta`, the value of the tail part is not uniform (if the hardware writes 1 to `ta` tail policy), but it is semantically consistent with tail agnostic. So I don't think the question I raised is a problem. I don't know if I missed anything in the above analysis, feel free to point it out.

lhtin, 2022-11-11:
> so I question whether the intrinsics API needs to support vector operands with EMUL != 1.

Agree with you.

> The only argument I see in favor of the current API is that the assembly programmer can "simulate" a (say) tail-undisturbed vmv.s.x with destination EMUL > 1 by careful register allocation:

Sorry, I didn't understand the key point here. Can you explain with more detail?

eopXD, 2023-10-24:
Want to confirm before closing this,

Interfaces of a greater LMUL are necessary for us to allow the first vector register of different LMUL grouping to be a target of `vmv.s.x`, since it is an instruction that is EMUL=1 regardless of the vtype setting.

Lack of this interface will bring inconvenience as users need to assemble the larger LMUL value themselves with `vset`/`vlmul_ext`.

I believe we can settle to what we have now.

closed 2023-10-25 by eopXD