riscv-non-isa/riscv-rvv-intrinsic-doc #178: Vector floating-point moves translated with strided loads

mp-17, 2022-11-22:
In my program, vector move instruction instrinsics, like `vfmv_v_f`, are translated to strided loads with `stride == 0`. 

Is there a way to tell (or force) the compiler to use scalar loads and then `scalar->vector` moves? Some decoupled architectures would benefit from this, especially if the memory hierarchy is asymmetric.

Example:
`vfloat64m1_t foo = vfmv_v_f_f64m1(1.389, VLMAX)`

Pseudo-ASM:
`fld f2, (@addr)`
`vfmv.v.f v2, f2`

Thanks a lot!

kito-cheng, 2022-11-22:
I could imagine such code gen strategy can benefit for specific core implemention, but that's compiler code gen issue, you could report that in llvm or gcc, this spec only specify the semantic of those intrinsic, and didn't specify the implemention :)

mp-17, 2022-11-22:
Thanks a lot to both of you, I will ask the LLVM community! ;-)