riscv-non-isa/riscv-rvv-intrinsic-doc #180: Exposure of vlenb missing for vread_csr

eopXD, 2022-12-27:
A frequently asked question about the API is why aren't vlenb available. It should also be something exposed by the intrinsic, this issue is a placeholder of this and we should resolve it.

tingsi5, 2022-12-27:
Is it possible to take `vl` and `vtype` into consdieration as well?

nick-knight, 2023-01-03:
> Is it possible to take vl and vtype into consdieration as well?

Can you describe the use-case you envision?

My opinion is we should maintain the current API where the `vl` and `vtype` CSRs are abstracted away from the C programmer. In the current API, the user specifies the desired semantics with the `size_t vl` argument and suffixes like `i64m8_tumu` in the intrinsics names, leaving the compiler to generate the necessary CSR accesses. Since the compiler has the freedom to reschedule code (as long as it preserves semantics), the C programmer will generally not know where, in the C program order, these accesses will be generated, so it's hard for me to envision how a programmer would use this mechanism usefully.

Also, adding more visibility into `vl` and `vtype` would take us down a similar path to C `fenv`, which is something we've been trying to avoid from the beginning of this project. I'm happy to go into more details.

`vlenb` is a different scenario because it is read-only (moreover, a "design-time constant"). So the underlying CSR read can be safely generated independently of surrounding (vector) code-gen

tingsi5, 2023-01-04:
Now I understand. Thank you for explaining me the reason why `vl` and `vtype` are not included from the beginning. I asked because as an user, it's unclear for me why read-only CSRs are excluded from the read API and thought if we're going to add `vlenb`, then probably we can take `vl` and `vtype` as well (I expected those CSRs are exposed intentionlly for read and should be included in the API).

If you have time I'd like to know the story of `fenv`. I googled it but got only the documentation.

closed 2023-01-10 by eopXD