riscv-non-isa/riscv-rvv-intrinsic-doc #190: Privileged instructions generated by GCC and rvv-instrisics

mariusmm, 2023-01-23:
Executing binary code generated by GCC using rvv-intrisics triggers in QEMU triggers an illegal instruction for vsetvli instruction.

It seems that this instruction among others are priviled instructions, thus cannot be executed by user executables.

Is something wrong with my toolchain or I'm missing some point about using RVV extension in linux environment?

nick-knight, 2023-01-23:
`vsetvli` and its relatives are unprivileged. 

Without full details to reproduce it's hard to say anything with certainty. But it sounds like your environment isn't setting `mstatus.VS` (privileged CSR) before your code is being executed.

I think your question should be redirected at the Linux kernel.

mariusmm, 2023-01-23:
Thank you for your answer. 

I've just compiled some intrinsic example and tried on QEMU with no-success. I see a bunch of 'vsetvli' instructions on the objdump that triggers the illegal instruction error.

Thank you

closed 2023-10-24 by eopXD