riscv-non-isa/riscv-rvv-intrinsic-doc #199: Question about "If XLEN < SEW, the value from the scalar register is sign-extended to SEW bits."

jinma-work, 2023-02-20:
hi，
"If XLEN < SEW, the value from the scalar register is sign-extended to SEW bits" is described in the [documen](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/5818330c298306714fefb705a524ddf3b9b50125/rvv-intrinsic-rfc.md#scalar-in-vector-operations)t, and I have some questions about it:

The function ```int64_t vmv_x_s_i64m1_i64 (vint64m1_t a);``` on RV32, that is, XLEN = 32,SEW = 64, has two situations:

1. According to the description "If XLEN < SEW, the value from the scalar register is sign-extended to SEW bits", only the lower 32 bits of ```v1``` register should be intercepted into a 32-bit scalar register ```a0```, and then ```a0``` should be signed and extended to 64 bits (```a1:a0```).

2. According to my understanding, the lower 32 bits of ```v1``` register should be copied to a 32-bit scalar register ```a0```, then the v1 register should be shifted to the right by 32 bits to generate vector register ```v2```, then the lower 32 bits of v2 should be copied to a 32-bit scalar register ```a1```, and finally 64-bit data (```a1:a0```)

I think 2 is more appropriate, because 1 will discard the high 32-bit data of ```v1```, but that "If XLEN < SEW, the value from the scalar register is sign-extended to SEW bits" makes me confused. Can someone answer for me? 
Thank you very much!

topperc, 2023-02-20:
2 is what clang implements. I don’t think 1 is a good idea. The only other option I see is making the intrinsic invalid for RV32.

jinma-work, 2023-02-20:
@topperc  OK, thank you very much for your reply. However, the document does not seem to clearly explain what should be done in this situation, and should it be clearly stipulated in the document? Avoid some doubts.

nick-knight, 2023-02-20:
The underlying instructions have different semantics for XLEN = 32 and XLEN = 64. You could argue they are different instructions. We could consider a similar approach to the  Zb* and Zk* intrinsics, and define `rv32_` and `rv64_` versions. Or we could use the same intrinsics name, but use [(proposed) `intxlen_t`](https://github.com/riscv-non-isa/riscv-c-api-doc/pull/14) to achieve the compile-time polymorphism. Or we could declare that the 1.0 milestone of this project only addresses the XLEN = 64 case, and kick the can down the road. We agreed a while back that approach (2) is a reasonable compromise, even if the behavior doesn't correspond to a single instruction. 

The sentence you're confused about was copy/pasted from [Section 10.1 of the V-extension spec](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#sec-arithmetic-encoding). It's specifically talking about the scalar input operands to certain vector arithmetic instructions. The instruction `vmv.x.s` is defined later, in [Section 16.1](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#161-integer-scalar-move-instructions). This instruction has a scalar output (vs. input) operand, so the sentence you quoted doesn't apply. Feel free to submit a PR if you think it needs a clarification.

jinma-work, 2023-02-20:
Thank you for your detailed reply.

I have no doubts about [Section 16.1](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#161-integer-scalar-move-instructions), just a little confused about the specific behavior of the function "X" on rv32. Although it is implemented according to approach (2) on clang, it is not explained in the document, which makes it easy for me to think that it is actually carried out according to approach (1).

I'm just suggesting :）

nick-knight, 2023-02-20:
@topperc what are we doing now for v*ei64.v on RV32?

topperc, 2023-02-20:
> @topperc what are we doing now for v*ei64.v on RV32?

For most `.vx` instructions we will splat the scalar into a vector using separate instructions then use the .vv form. Worst case 2 stores and a strided load with x0 stride. We do try to use the .vx instruction if the scalar is sign extended.

We have a special cases for slide1up/slide1down and vmv.s.x.

vmv.x.s uses a slidedown and a vmv.x.s to extract all 64 bits.

nick-knight, 2023-02-20:
Thanks, but I was asking about this:

> The V extension supports all vector load and store instructions (Section [Vector Loads and Stores](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#sec-vector-memory)), except the V extension does not support EEW=64 for index values when XLEN=32.

topperc, 2023-02-20:
> Thanks, but I was asking about this:
> 
> > The V extension supports all vector load and store instructions (Section [Vector Loads and Stores](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#sec-vector-memory)), except the V extension does not support EEW=64 for index values when XLEN=32.

I think the ei64 indexed load/store intrinsics are not available on RV32.

nick-knight, 2023-02-20:
And do we also disable vector floating-point intrinsics when the associated scalar floating-point extension is not available?

topperc, 2023-02-20:
> And do we also disable vector floating-point intrinsics when the associated scalar floating-point extension is not available?

I think we disable the floating point intrinsics based on which V/Zve*/Zvfh extensions are enabled. The FP Zve* and Zvfh extensions can’t be enabled without scalar extensions also being enabled. And V implies F and D.

nick-knight, 2023-02-20:
Then it seems there is also an error (in addition to a typo) in

> There is the same issue for `vmv.x.s`, `vmv.s.x`, `vfmv.f.s`, `vfmv.s.f`, `vslide1up.vx`, `vfslide1up.vf`, `vslide1down.vx`, and `vfslide1down.vx`. Use `SEW` to encode the scalar type.

The floating-point instructions should not be mentioned, do you agree?

nick-knight, 2023-02-20:
I've attempted to clarify the confusion by rewriting this section: please review https://github.com/riscv-non-isa/rvv-intrinsic-doc/pull/201.

PR #201 (2023-02-22): Clarify semantics for scalar operands
The existing text was phrased in a way that may cause confusion in the case of instructions with scalar outputs, like `vmv.x.s`. 

There was also some redundancy, a typo in one of the assembly mnemonics, and failure to mention that the scalar operands are sometimes zero- (vs. sign-) extended.

I've rewritten the section to address these things, and close #199.
closed 2023-02-22 by eopXD