riscv-non-isa/riscv-rvv-intrinsic-doc #203: are there intrinsics for the whole register load/store instructions?

fuhle044, 2023-02-28:
Hi,

I'm looking for these intrinsics and I am not finding them. Do we have them or are they in an unexpected place?

-rich

nick-knight, 2023-03-16:
I stumbled across this issue while I was looking for intrinsics for whole-register moves (`vmv<nr>.v`), which also don't exist. Regarding these moves, @eopXD wrote:

> We currently have = with semantic of a [whole vector register move](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#166-whole-vector-register-move) (which is also vmv.v.v of VL=VLMAX).

https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/74#issuecomment-1202231257

I don't think this has been documented yet.

I've only given it about five minutes of thought, but I think this is a reasonable API design decision.

Regarding your question about whole-register loads/stores, I don't recall if the omission was intentional. I'm struggling for a use-case as an intrinsics programmer. The only use-case I see is context save/restore code, but we've already agreed to ignore this use-case in our API design (e.g., we already don't expose R-type `vsetvl`).

nick-knight, 2023-03-21:
Hi Rich, 

I brought this up at this week's task group meeting. The conclusion is that the API intentionally omits whole-register loads, stores, and moves. The rationale is that the usual loads/stores/moves provide the same functionality, and a compiler could instead generate whole-register versions if it thought it would be profitable. If a compelling use-case arises in the future, we could introduce new intrinsics in a backwards-compatible way.

Best,
Nick

closed 2023-10-24 by eopXD