riscv-non-isa/riscv-rvv-intrinsic-doc #205: What's the meaning of new_vl in Unit-stride Fault-Only-First Loads Functions?

zfcfuture, 2023-03-01:
Hello, thank you very much for the riscv-vector api interface you provided.
But I have some doubts when using  Unit-stride Fault-Only-First Loads Functions.
I don't know what's the meaning of `size_t *new_vl` and how to use it. And can you tell me how to use this api?
Hope to get your reply as soon as possible

eopXD, 2023-03-01:
[strncpy](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/master/examples/rvv_strncpy.c) is a case where the instruction comes in handy.

topperc, 2023-03-01:
The fault-only-first-load intrinsic expands to a `vleff` instruction and a `csrr` to read the `vl` register after the `vleff`. `new_vl1 is where to store the result from the `csrr`

zfcfuture, 2023-03-01:
> The fault-only-first-load intrinsic expands to a `vleff` instruction and a `csrr` to read the `vl` register after the `vleff`. `new_vl` is where to store the result from the `csrr`

thank you! i understand

zfcfuture, 2023-03-01:
> [strncpy](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/master/examples/rvv_strncpy.c) is a case where the instruction comes in handy.

thank you!

closed 2023-03-01 by zfcfuture