riscv-non-isa/riscv-rvv-intrinsic-doc #211: How to deal with some unsupported interfaces on rv32?

mar-jin, 2023-03-06:
According to my understanding, there are some interfaces that are not supported on rv32, such as ```void __riscv_vluxseg3ei64_v_f64m1_m (vfloat64m1_t *v0, vfloat64m1_t *v1, vfloat64m1_t *v2, vbool64_t mask, const float64_t *base, vuint64m1_t bindex, size_t vl);```, which are mainly [Vector Indexed Segment Load Functions](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/master/auto-generated/intrinsic_funcs) interfaces  containing ```vuint64m1_t bindex```. I think we should add error reporting information to explain this problem to users. Or is there a better way?

eopXD, 2023-03-13:
Nice catch, I think this is more of a compiler issue rather than a spec. related issue. May you help submit a bug under LLVM so we can track this?

eopXD, 2023-10-24:
This is a matter of compiler implementation and not a spec issue.

On the other hand, now in LLVM we do guard uses to RVV values of element width=64 in semantic analysis.

closed 2023-10-24 by eopXD