riscv-non-isa/riscv-rvv-intrinsic-doc #232: question about naming of overloaded policy load functions

kuanlinchentw, 2023-06-01:
Hi,
Why _mu series load intrinsics cannot be overloaded without type in the function name?

In [vector_loads_and_stores_functions](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/master/auto-generated/policy_funcs/overloaded_intrinsic_funcs/01_vector_loads_and_stores_functions.md), __riscv_vle16_v_f16mf4_mu, __riscv_vle16_v_f16mf2_mu, etc., are with type in the function name.
But __riscv_vle16_tumu,__riscv_vle16_tum, etc., are without types in the function name.

Thanks.

eopXD, 2023-06-01:
Thank you for posting this, I think you have found a miss in the specification (and the generated unit tests). I have just checked the latest upstream LLVM compiler and it is able to compile the following.

```
vfloat16mf2_t test_vle16_mu(vbool32_t mask, vfloat16mf2_t maskedoff, const _Float16 *base, size_t vl) {
  return __riscv_vle16_mu(mask, maskedoff, base, vl);
}
```

I will submit a patch to fix this. Thanks again for checking.

kito-cheng, 2023-06-01:
Seem like GCC has honor the spec...:P

https://godbolt.org/z/xP5e4csnv

kuanlinchentw, 2023-06-01:
BTW, __riscv_viota_m_u8mf8_mu is the same.

zhongjuzhe, 2023-06-01:
> Seem like GCC has honor the spec...:P
> 
> https://godbolt.org/z/xP5e4csnv

Yeah, GCC has strictly followed SPEC:

bool can_be_overloaded_p (enum predication_type_index pred) const override
  {
    if (STORE_P || LST_TYPE == LST_INDEXED)
      return true;
    return pred != PRED_TYPE_none && pred != PRED_TYPE_mu;
  }

....

It seems that I need to remove pred != PRED_TYPE_mu;

PR #233 (2023-06-03): Add missing `_mu` variants for some overloaded intrinsics
Resolves #232
commit dbd27049 (2023-06-03): Add missing `_mu` variant for overloaded functions

Resolves #232
closed 2023-06-03 by eopXD