riscv-non-isa/riscv-rvv-intrinsic-doc #241: Question: What is the status of rvv intrinsic API 1.0 release

fanghuaqi, 2023-07-04:
Hello, everyone,

I am wondering what is current status about rvv instrinsic API doc, when will intrinsic API v1.0 will be released, is there any timeline on it? And I saw gcc 13 release support rvv 0.11 release, and I also saw https://gcc.gnu.org/bugzilla/show_bug.cgi?id=110448 reported a incompatiable issue with gcc master branch, I am curious about these questions:

1. When will rvv intrinsic API will be freezed, and supported status in llvm and gcc?
2. In this master branch, there is a [rvv-intrinsic-generator](https://github.com/riscv-non-isa/rvv-intrinsic-doc/tree/master/rvv-intrinsic-generator) test suite, is this can be treat as golden test suites for rvv intrinsic API compatiable suite, so we can use this suite to evaluate different version of gcc/clang support status?
3. If 2's answer is yes, is there any steps to evaluate on gcc and clang toolchain?

Thanks

asb, 2023-07-06:
Just noting that LLVM 17 is due to branch on July 25th (final release due 6 weeks after, with further changes becoming progressively harder to justify cherry-picking as that date gets closer). If there are any disruptive changes coming down the pipe, it might be worth thinking how best to minimise disruption to end users given that release schedule.

eopXD, 2023-07-07:
The remaining functional change left before the RVV v1.0 C intrinsics is to add floating-point variants that controls the rounding mode [0]. I believe we are done on all disruptive (incompatible) changes planned for v1.0.

[0] D154628 [[1/8][RISCV] Add rounding mode control variant for vfsub, vfrsub](https://reviews.llvm.org/D154628)

fanghuaqi, 2023-07-07:
> 

Hi @eopXD , thanks for the information, so I can understand it like this:

1. current https://github.com/riscv-non-isa/rvv-intrinsic-doc master branch contains the v1.0 intrinisc API, no more updates will be included?
2. llvm main branch will be fully compatiable rvv v1.0 c intrinsic when this D154628 merged?
3. what is the status for gcc master branch, is all the v1.0 c intrinisc APIs implemented? 
4. Can I use the test suite provided in this repo as compatiablity test suite for rvv v1.0 intrinsic API?

Thanks

eopXD, 2023-07-07:
Yes LLVM will support v1.0 after the patch-set of 

> 1. current https://github.com/riscv-non-isa/rvv-intrinsic-doc master branch contains the v1.0 intrinisc API, no more updates will be included?

We still have #226  yet to be confirmed in implementation and merged into master. Then we will have about one month to polish the current documentations.

> 2. llvm main branch will be fully compatiable rvv v1.0 c intrinsic when this D154628 merged?

Yes LLVM will support v1.0 after the patch-set of D154628.

> 3. what is the status for gcc master branch, is all the v1.0 c intrinisc APIs implemented?

I will defer this question to @kito-cheng.

> 4. Can I use the test suite provided in this repo as compatiablity test suite for rvv v1.0 intrinsic API?
> 

Of course, please try it out and you are welcome to report issues.

fanghuaqi, 2023-07-07:
Thank you @eopXD for your explanation.

zhongjuzhe, 2023-07-07:
> Hi @eopXD , thanks for the information, so I can understand it like this:
>
> 1. current https://github.com/riscv-non-isa/rvv-intrinsic-doc master branch contains the v1.0 intrinisc API, no more [...]

We are working on it. Hopefully all APi will be done in this month.
We have already supported for fixed-point api with rounding mode:
https://godbolt.org/z/a3nbfor3r

But floating-point is not yet since dynamic rounding mode needs some tricks but won't be the problem.

fanghuaqi, 2023-08-21:
Hi @zhongjuzhe ， is there timeline when will rvv intrinsic 1.0 be supported in gcc master branch, we checked in master, still not work for this repo's intrinsic api testing, and will the changes be synced to https://gcc.gnu.org/git/?p=gcc.git;a=shortlog;h=refs/vendors/riscv/heads/gcc-13-with-riscv-opts branch?

Thanks

Incarnation-p-lee, 2023-08-21:
Hi @fanghuaqi 
Thanks for the question, we (GCC) are very close to supporting all the API of rvv intrinsic 1.0, I bet there are a few patches under review and we are also testing all the intrinsic API generated by this repo.

For the question to gcc-13-with-riscv-opts, I think we need the call from kito.

Currently, I am working on the rest intrinsic API implementation in master, as well as the full test. Please feel free to let me know if any questions or concerns.

Thanks

fanghuaqi, 2023-08-21:
Hi @Incarnation-p-lee , thank you for your reply, do you think it will be possible to be done in 2 or 3 weeks, are all the rest patches can be found in https://patchwork.sourceware.org/project/gcc/list/?q=risc-v . We want to cherry-pick these rvv intrinsic api 1.0 support to gcc13 release version, and we found this riscv sync branch https://gcc.gnu.org/git/?p=gcc.git;a=shortlog;h=refs/vendors/riscv/heads/gcc-13-with-riscv-opts , it contains some rvv patches on gcc 13 release branch, so I am wondering whether this branch will sync all the rvv intrinsic api changes done by you, could @kito-cheng give me some hints on this, thanks.

Thank you

kito-cheng, 2023-08-21:
[gcc-13-with-riscv-opts](https://gcc.gnu.org/git/?p=gcc.git;a=shortlog;h=refs/vendors/riscv/heads/gcc-13-with-riscv-opts) is mostly maintained by Ventana folks, and they will pick most RISC-V related patch to that branch, so no guarantee from me :P but I think they will regularly run test on that branch, so you can report bug to gcc mailing list and cc jeffreyalaw@gmail.com and me (if it related to intrinsic)

fanghuaqi, 2023-08-22:
Thanks @kito-cheng and @Incarnation-p-lee for your answers, helps a lot.

fanghuaqi, 2023-09-01:
Hello @Incarnation-p-lee , is current gcc master branch ready with rvv 1.0 intrinsic API support now? I see almost all of the patches related to rvv intrinisc list in https://patchwork.sourceware.org/project/gcc/list/?q=risc-v are merged.

Incarnation-p-lee, 2023-09-01:
Almost all API(s) of RVV 1,0 intrinsic are supported AFAIK, we are validating if there are something like bugs/missed in the upstream of GCC.

fanghuaqi, 2023-09-01:
> Almost all API(s) of RVV 1,0 intrinsic are supported AFAIK, we are validating if there are something like bugs/missed in the upstream of GCC.

Thank you @Incarnation-p-lee , I just checked with latest gcc master branch, using script located in `https://github.com/riscv-mcu/rvv-intrinsic-doc/blob/nuclei/main/rvv-intrinsic-generator/check_api.sh#L9`

This change need to be made for gcc/clang run https://github.com/riscv-mcu/rvv-intrinsic-doc/commit/4f4ca59eb3192703f736cae9dcd773659087978b

And find there are still some api test suite not working, please check this fail patterns [gcc_failtxt_20230901_1616.zip](https://github.com/riscv-non-isa/rvv-intrinsic-doc/files/12495920/gcc_failtxt_20230901_1616.zip):

[image]

zhongjuzhe, 2023-09-01:
All fails are caused by:
warning: ABI for the vector type is currently in experimental stage and may changes in the upcoming version of GCC. [-Wpsabi]

These are not GCC bugs. Vector ABI is ratified,  you should use -Wno-psabi to fix the warnings.

***@***.***
 
From: Huaqi Fang
Date: 2023-09-01 16:23
To: riscv-non-isa/rvv-intrinsic-doc
CC: Ju-Zhe Zhong; Mention
Subject: Re: [riscv-non-isa/rvv-intrinsic-doc] Question: What is the status of rvv intrinsic API 1.0 release (Issue #241)
Almost all API(s) of RVV 1,0 intrinsic are supported AFAIK, we are validating if there are something like bugs/missed in the upstream of GCC.
Thank you @Incarnation-p-lee , I just checked with latest gcc master branch, using script located in https://github.com/riscv-mcu/rvv-intrinsic-doc/blob/nuclei/main/rvv-intrinsic-generator/check_api.sh#L9
This change need to be made for gcc/clang run ***@***.***
And find there are still some api test suite not working, please check this fail patterns gcc_failtxt_20230901_1616.zip:

fanghuaqi, 2023-09-01:
Hi @zhongjuzhe , when I passed `-Wno-psabi`, there are still many errors such as 

[image]

For complete fail patterns, check 
[gcc_failtxt_20230901_1635.zip](https://github.com/riscv-non-isa/rvv-intrinsic-doc/files/12496075/gcc_failtxt_20230901_1635.zip)

zhongjuzhe, 2023-09-01:
-march=rv64gcv_zvfh

***@***.***
 
From: Huaqi Fang
Date: 2023-09-01 16:40
To: riscv-non-isa/rvv-intrinsic-doc
CC: Ju-Zhe Zhong; Mention
Subject: Re: [riscv-non-isa/rvv-intrinsic-doc] Question: What is the status of rvv intrinsic API 1.0 release (Issue #241)
Hi @zhongjuzhe , when I passed -Wno-psabi, there are still many errors such as
For complete fail patterns, check
gcc_failtxt_20230901_1635.zip

fanghuaqi, 2023-09-01:
Hi @zhongjuzhe , just changed it, but seems not working for me.

fanghuaqi, 2023-09-01:
Same test suites are executed on clang, all four types of api testing are passed.

[image]

Incarnation-p-lee, 2023-09-01:
Suggest reference this [CI](https://github.com/Incarnation-p-lee/rvv-intrinsic-doc/actions/workflows/gnu-gcc-daily-build.yml), which covered the most popular arches combination suggested by Juzhe.

Let me double-check about the fp16 parts.

fanghuaqi, 2023-09-01:
> Suggest reference this [CI](https://github.com/Incarnation-p-lee/rvv-intrinsic-doc/actions/workflows/gnu-gcc-daily-build.yml), which covered the most popular arches combination suggested by Juzhe.
> 
> Let me double-check about the fp16 parts.

Ok, thanks, I found you are using `rvv-intrinsic-doc/gnu-auto-generated` cases, what are differences between this repo's auto-generated cases?

Thank you

fanghuaqi, 2023-09-01:
And by the way, will gcc support all four types rvv intrinsic API list in this pdf https://github.com/riscv-non-isa/rvv-intrinsic-doc/releases/tag/draft-20230901-1e1ea8f13222129f52bb93cf33e48c92e34811d0

[image]

From my understanding, I choosed to run this **four targets**:

- make COMPILER=riscv64-unknown-elf-clang run-api-testing
- make COMPILER=riscv64-unknown-elf-clang run-overloaded-api-testing
- make COMPILER=riscv64-unknown-elf-clang run-policy-api-testing
- make COMPILER=riscv64-unknown-elf-clang run-policy-overloaded-api-testing

Incarnation-p-lee, 2023-09-01:
> Ok, thanks, I found you are using `rvv-intrinsic-doc/gnu-auto-generated` cases, what are differences between this repo's auto-generated cases?
>
> Thank you

it is base on this repo but has some additional check in the GNU style.

fanghuaqi, 2023-09-04:
> Suggest reference this [CI](https://github.com/Incarnation-p-lee/rvv-intrinsic-doc/actions/workflows/gnu-gcc-daily-build.yml), which covered the most popular arches combination suggested by Juzhe.
> 
> Let me double-check about the fp16 parts.

Hi @Incarnation-p-lee , is there any new feedback on this different testing results on this repo and your maintained repo, my test result can be found in [gcc_failtxt_20230901_1635.zip](https://github.com/riscv-non-isa/rvv-intrinsic-doc/files/12496075/gcc_failtxt_20230901_1635.zip), thanks.

Incarnation-p-lee, 2023-09-04:
Hi, we checked the diff by a script and find only VRGATHEREI16 lacks FP16 support. We are preparing the patch for this, and meanwhile, trigger a fully test on auto-generated/api-testing to ensure consistency.

fanghuaqi, 2023-09-04:
> Hi, we checked the diff by a script and find only VRGATHEREI16 lacks FP16 support. We are preparing the patch for this, and meanwhile, trigger a fully test on auto-generated/api-testing to ensure consistency.

Hi @Incarnation-p-lee ， thanks, then will gcc support these four types of API, see https://github.com/riscv-non-isa/rvv-intrinsic-doc/issues/241#issuecomment-1702420925

fanghuaqi, 2023-09-04:
It seems, except VRGATHEREI16 fp support is not yet ready, overloaded type api is not supported by checking `fail_gcc_policy-overloaded-api-testing.txt` and `fail_gcc_overloaded-api-testing.txt` in this the attached zip file.

fanghuaqi, 2023-09-04:
Hi @Incarnation-p-lee , I have pulled latest gcc commit, now `api-testing` and `policy-api-testing` are passed, but overloaded api are still failing, does gcc has plan to support overloaded api or not, thanks.

Incarnation-p-lee, 2023-09-04:
Thanks @fanghuaqi for reporting this, let me double check the overloaded part and keep you posted.

fanghuaqi, 2023-09-05:
> Hi @fanghuaqi,
> 
> Double checked the intrinisc API(s) are fully supported since this commit [1e1ea8f](https://github.com/riscv-non-isa/rvv-intrinsic-doc/commit/1e1ea8f13222129f52bb93cf33e48c92e34811d0) of rvv-intrinsic-doc.
> 
> You can leverage this [CI](https://github.com/Incarnation-p-lee/rvv-intrinsic-doc/actions/runs/6080726902/job/16495127272) for reference.

Hi @Incarnation-p-lee , thank you, I will check the difference of the overloaded part.

fanghuaqi, 2023-09-06:
> 

Hi @Incarnation-p-lee , I found that for the overloaded part, your script use g++ to compile, when use gcc it will fail, this is what I observed using gcc in this repo, any reason why I need to use g++ to use overloaded api, thanks.

see https://godbolt.org/z/P4P58s7cd

[image]

zhongjuzhe, 2023-09-06:
overloaded is C++ standard we should not support overload on C. Here is the simple example:
https://godbolt.org/z/va8To88bK

fanghuaqi, 2023-09-06:
> 

OK, thanks @zhongjuzhe for your explaination.

dzaima, 2023-09-06:
> https://godbolt.org/z/va8To88bK

@zhongjuzhe the clang error there isn't because the overloaded rvv intrinsics aren't available, but because of the collided test function names. in e.g. https://godbolt.org/z/M76Wj63zP you can see that clang does support the overloaded intrinsics even in C.

zhongjuzhe, 2023-09-06:
It doesn't make sense the rvv intrinsics can be overloaded but the others can't be.
I don't think this is GCC's issue.

More informations about overloaded, plz read this:
https://stackoverflow.com/questions/3178342/compiling-a-c-program-with-gcc 

***@***.***
 
From: dzaima
Date: 2023-09-06 22:35
To: riscv-non-isa/rvv-intrinsic-doc
CC: Ju-Zhe Zhong; Mention
Subject: Re: [riscv-non-isa/rvv-intrinsic-doc] Question: What is the status of rvv intrinsic API 1.0 release (Issue #241)
https://godbolt.org/z/va8To88bK
@zhongjuzhe the clang error there isn't because the overloaded rvv intrinsics aren't available, but because of the collided test function names. in e.g. https://godbolt.org/z/M76Wj63zP you can see that clang does support the overloaded intrinsics even in C.

kito-cheng, 2023-09-06:
clang and gcc both has capability to support overloaded function on C, also rvv intrinsic didn’t disallow that for c

zhongjuzhe, 2023-09-06:
Ok. Sounds reasonable. We can do that later.
We can support that if I find some one can help (I won't dedicate myself on intrinsics, I am only working on auto-vectorization).

***@***.***
 
From: dzaima
Date: 2023-09-06 22:47
To: riscv-non-isa/rvv-intrinsic-doc
CC: Ju-Zhe Zhong; Mention
Subject: Re: [riscv-non-isa/rvv-intrinsic-doc] Question: What is the status of rvv intrinsic API 1.0 release (Issue #241)
@zhongjuzhe You don't need overloaded functions here, it can be done in pure C with _Generic, e.g. https://godbolt.org/z/KhfKE48nK. Given that, I don't think C vs C++ really matters here.

Incarnation-p-lee, 2023-09-07:
Thanks all, will take care of the overloaded intrinsic API support in gcc soon.

zhongjuzhe, 2023-09-07:
> Thanks all, will take care of the overloaded intrinsic API support in gcc soon.

To support that, I think you may need to add a new targethook 'resolve_overloaded_builtin'.
You can reference how aarch64 support it.

fanghuaqi, 2023-09-07:
This repo is adding a `vcreate` api, see https://github.com/riscv-non-isa/rvv-intrinsic-doc/pull/256, which is not yet supported by gcc master branch.

eopXD, 2023-10-24:
Can this be closed now?

closed 2023-10-25 by eopXD

zhongjuzhe, 2023-11-06:
All C/C++ RVV intrinsics has been supported on GCC.
And @xl0714 supported C compiler C++ overloaded intrinsic on GCC:
https://github.com/riscv-non-isa/rvv-intrinsic-doc/pull/294
https://gcc.gnu.org/git/?p=gcc.git;a=commit;h=1a55724f7870719d51a7acc0ccada254fcfd1cda