riscv-non-isa/riscv-rvv-intrinsic-doc #247: Question regarding __riscv_vsetvl mask/tail policy

OMaghiarIMG, 2023-07-21:
Hello,

I was looking at an example in which we use _tu intrinsics:
[image]

The first __riscv_vsetvl translates to vsetvli ta,mu, then __riscv_vle32 adds another vsetvl to set the policy to tu,ma.
Would it be possible to have a single vsetvl with tu,ma since the beginning?

Would having _mu/_tu variants of the __riscv_vsetvl intrinsic make sense?

zhongjuzhe, 2023-07-21:
No, I don't think we need to add _mu/_tu variants of __riscv_vsetvl intrinsic.

__riscv_vsetvl intrinsic is defined as any tail policy and any mask policy.

Compiler should be able to well optimize vsetvlis.

Compiler can well handle your case, no redundant vsetvli:

https://godbolt.org/z/jMra9dMKG

OMaghiarIMG, 2023-07-21:
Thanks, yes this issue was present on my local clang16, didn't look at gcc.
Here's an example with a mask setting instruction, gcc sets tu,ma since the beginning, clang does a switch to ta and then back to tu.
https://compiler-explorer.com/z/TE5fzEh7j

zhongjuzhe, 2023-07-21:
I think you should try trunk Clang.

The intrinsic has significantly changed recently and will be frozen soon.
Clang16 didn't support the latest rvv intrinsics.