riscv-non-isa/riscv-rvv-intrinsic-doc #258: question about vm

unicornx, 2023-08-21:
https://github.com/riscv-non-isa/rvv-intrinsic-doc/releases/tag/draft-20230811-a810011071e9e2f630450f01f5fdc9a9ccc3e3be

6.1 Policy naming scheme, in the example
```
// vm=1, vta=0
vint32m1_t __riscv_vadd_vv_i32m1_tu(vint32m1_t maskedoff, vint32m1_t op1,vint32m1_t op2, size_t vl);
```

To my understand, when vm=1, means unmasked vector operation, why do we need provide `maskedoff` as the first parameter?

And for the last four examples:
```
// vm=0, vta=1, vma=1
vint32m1_t __riscv_vadd_vv_i32m1_m(vbool32_t mask, vint32m1_t op1,vint32m1_t op2, size_t vl);
// vm=0, vta=0, vma=1
vint32m1_t __riscv_vadd_vv_i32m1_tum(vbool32_t mask, vint32m1_t maskedoff,vint32m1_t op1, vint32m1_t op2, size_t vl);
// vm=0, vta=1, vma=0
vint32m1_t __riscv_vadd_vv_i32m1_mu(vbool32_t mask, vint32m1_t maskedoff,vint32m1_t op1, vint32m1_t op2, size_t vl);
// vm=0, vta=1, vma=1
vint32m1_t __riscv_vadd_vv_i32m1_tumu(vbool32_t mask, vint32m1_t maskedoff,vint32m1_t op1, vint32m1_t op2,size_t vl);
```

To my understand, when vm=0, which means we enable mask, why we need  `vbool32_t mask`, what it is used for, and I think `vint32m1_t maskedoff` should be required and it is enough to specify which one is masked-off.

I wonder if there's some misunderstanding for me, please feel free correct me and point it out, thanks.

topperc, 2023-08-21:
The `maskeoff` operand for `_tu` intrinsics provides the tail elements. The naming could be improved.

For the vm=0 instructions, `vbool32_t mask` is the mask to put in the `v0` register. The `maskedoff` arguement provides the undisturbed value for the tail and/or the masked off elements. For the `_m` intrinsic there is no `maskedoff` argument because the intrinsic doesn't guarantee any value for the masked off or tail elements after the intrinsic. It will use a `ta` and `ma` policy and the compiler may pick any available register to hold the result.

eopXD, 2023-10-23:
This is recently addressed in #271.

closed 2023-10-23 by eopXD