riscv-non-isa/riscv-rvv-intrinsic-doc #262: Does the project have absolute value functions for int and uint integer vectors?

rvvstuding, 2023-08-29:
if not, how do I implement them like Floating-Point Absolute Value Functions?

dzaima, 2023-08-29:
This is probably the best option:
```c
vbool16_t tmp = __riscv_vmslt_vx_i32m2_b16(x, 0, vl);
vint32m2_t res = __riscv_vneg_v_i32m2_mu(tmp, x, x, vl);
```
that is, conditionally negate elements that are less than 0.

rvvstuding, 2023-08-30:
Thank you for your answer, by the way, will there be intrinsics functions like ARM vabs_s8 and x86 _mm_abs_epi8 in the future?

topperc, 2023-08-30:
> Thank you for your answer, by the way, will there be intrinsics functions like ARM vabs_s8 and x86 _mm_abs_epi8 in the future?

Intrinsics usually represent a single instruction. ARM and X86 have sing instruction abs. RISC-V does not.

You could also do vmax with the value and a negated version of the value. Depending on hardware that might chain better, but have higher register pressure for larger LMUL.

nick-knight, 2023-08-30:
> You could also do vmax with the value and a negated version of the value. Depending on hardware that might chain better, but have higher register pressure for larger LMUL.

This is the second option @dzaima mentioned. But they added that in an edit to their original post, so if you were coming here from your email you may have missed it. Agreed about the tradeoffs.