riscv-non-isa/riscv-rvv-intrinsic-doc #263: Question about rvv intrinsic version

rvvstuding, 2023-08-30:
Hello, everyone,
  I saw gcc 13 release support rvv 0.11 release, and I want to know how to find the version of rvv intrinsic version in my gcc(gcc version 12.0.1 rvv-next branch). And I wonder which is more stable and recommended between the my version and rvv 0.11 release.
Thanks

zhongjuzhe, 2023-09-04:
I would suggest to use master/trunk GCC which supports V1.0 intrinsics.
Besides, rvv-next branch is going to retire/remove since we no longer maintain rvv-next.

eopXD, 2023-09-04:
You may use the test macro to checkout the supported version of the compiler. On the other hand, I agree with @zhongjuzhe  that it is at best for you to start the developments on top of latest intrinsics since the old ones will be deprecated.

rvvstuding, 2023-09-04:
> You may use the test macro to checkout the supported version of the compiler. On the other hand, I agree with @zhongjuzhe that it is at best for you to start the developments on top of latest intrinsics since the old ones will be deprecated.

I found the `version_macro.adoc` file in the old documents, which exists `= Version Macro`, and I tried to use it, in the form of `int v = Version Macro` , but failed. Am I using it in a wrong way?

closed 2023-09-15 by rvvstuding