riscv-non-isa/riscv-rvv-intrinsic-doc #270: Question about Reinterpret Cast Conversion Functions

rvvstuding, 2023-09-11:
Hi  everyone,
    As far as I know about reinterpret Intrinsics, they can reinterpret vector between different type under the same SEW/LMUL, or between different SEW under the same LMUL. 
    1. I don't see the reinterpret between float type to different SEW float type, such as `vfloat16m1_t` to `vfloat32m1_t`
    2. If  I want to reinterpret `vint8m1_t `to `vuint16m1_t`, I can be achieved by using  `__riscv_vreinterpret_v_i16m1_u16m1(__riscv_vreinterpret_v_i8m1_i16m1(a))`. If this is right, How do I reinterpret `vfloat16m1_t `to `vint8m1_t`?
Thanks!

dzaima, 2023-09-11:
`f16`→`i8` can be `__riscv_vreinterpret_v_i16m1_i8m1(__riscv_vreinterpret_v_f16m1_i16m1(x))`.

`f16`→`f32` is a more unwieldy `__riscv_vreinterpret_v_u32m1_f32m1(__riscv_vreinterpret_v_u16m1_u32m1(__riscv_vreinterpret_v_f16m1_u16m1(x)))`, but given that there's not much use for it, I think it's fine.

nick-knight, 2023-09-11:
Correct: the current API doesn't form the full Cartesian product of conversions, but there should be a path between any pair of types with same EMUL.

rvvstuding, 2023-09-12:
Thanks for your reply, my problem is solved！

closed 2023-09-15 by rvvstuding