riscv-non-isa/riscv-rvv-intrinsic-doc #276: Question about intrinsic data types

GaryCAICHI, 2023-09-18:
To my opinion, currently rvv data types are defined under the restriction of ELEN=64, so data types like vint64mf2_t are illegal. The thing is, for those processors(T-Head C920, Sophon SG2042 etc.) which adopt 128-bit length vector registers, if I want to set SEW=64, using vint64m1_t means that only 64 bits of the 128 bits in the vector register are used. This results in a waste of spatial resources. Is this going to be improved or do I get something wrong? Thanks!

dzaima, 2023-09-18:
`ELEN` is the maximum width of an element; C920 with RVV1.0 and 128-bit vectors (`VLEN=128`) would still have `ELEN=64`, and a `vint64m1_t` would hold two 64-bit integers (and the same code, if ran on some other processor with 256-bit vector registers, would hold 4 64-bit ints).

Thus, `vint64m1_t` never wastes any space (`m1` types, by definition, are exactly one full vector register; `mf2` would be explicitly having a vector register whose upper half goes unutilized). The `mf2`/`mf4`/`mf8` types exist only to provide for the possibility of widening, but a 64-bit integer cannot be widened to anything (you'd need 128-bit elements for that, which don't exist in RVV1.0).

GaryCAICHI, 2023-09-18:
> `ELEN` is the maximum width of an element; C920 with RVV1.0 and 128-bit vectors (`VLEN=128`) would still have `ELEN=64`, and a `vint64m1_t` would hold two 64-bit integers (and the same code, if ran [...]

Thanks! Seems that I misunderstand the definition of ELEN, I'll check it out right away. But I may have further questions of how this works, shall I close the issue now and open a new issue or left this issue open?

dzaima, 2023-09-18:
I can't answer on what'd be the best format for more questions/issue format (I'm just a "passer-by" here); generally, though, ELEN is gonna be 64 on nearly all standard RVV1.0 implementations no matter the register width for the foreseeable future ("v" extension requires VLEN≥128 and supported EEW of 8,16,32,64).

eopXD, 2023-09-19:
You may find descriptions of data type availabilities under [Type System](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/eopc/edit-spec/doc/rvv-intrinsic-spec.adoc#type-system).
You may find descriptions of control number of elements to be processed under [Control of number of elements to be processed](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/eopc/edit-spec/doc/rvv-intrinsic-spec.adoc#control-of-number-of-elements-to-be-processed).

closed 2023-09-20 by GaryCAICHI