riscv-non-isa/riscv-rvv-intrinsic-doc #277: [Error] Conversion from vuint to vbool

Tameem97, 2023-09-18:
Hi,

For the following code,

```
int main() {

    uint8_t arr[32] = {1, 2, 7, 1, 3, 4, 5, 3, 1, 0, 1, 2, 4, 4, 9, 9, 1, 2, 7, 1, 3, 4, 5, 3, 1, 0, 1, 2, 4, 4, 9, 9};
    uint8_t m = 1;

    vuint8m1_t varr = __riscv_vle8_v_u8m1(arr, 32);
    vuint8m1_t vand_m = __riscv_vand_vx_u8m1(varr, m, 32);
    vbool8_t vmask = __riscv_vreinterpret_v_u8m1_b8(vand_m);

    //  ... 
    //  ...

    return 0;
}
```
\
 I am trying to use this function as shown in the intrinsics view by dzaima,
[image]

But I am getting this error,

```
$   riscv64-unknown-linux-gnu-gcc ./temp.c 
./temp.c: In function 'main':
./temp.c:18:22: warning: implicit declaration of function '__riscv_vreinterpret_v_u8m1_b8'; did you mean '__riscv_vreinterpret_v_u8m1_i8m1'? [-Wimplicit-function-declaration]
   18 |     vbool8_t vmask = __riscv_vreinterpret_v_u8m1_b8(vand_m);
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                      __riscv_vreinterpret_v_u8m1_i8m1
./temp.c:18:22: error: incompatible types when initializing type 'vbool8_t' using type 'int'
```
\
Maybe I am misusing the vreinterpret function, or missing something but it is not clear to me why this error is occurring.
\
Actually, I am trying to do this (C code equivalent)

```
    for (int n=0; ...) {
        m <<= 1;
        // To be vectorised
        for (int i=0; i < 32; i++) {
            if (arr[i] & m) {
                ...
            }
        }
    }

```
\
Thank you,

dzaima, 2023-09-18:
Also, you might want something like `__riscv_vmsne_vx_i8m1_b8(vand_m, 0, 32)` instead; the reinterpret will reinterpret (giving 8 bits per each int8 element here) whereas a comparison will give 1 mask bit per element.

Tameem97, 2023-09-18:
okay, I thought vreinterpret will also be 1-bit, I will then use the  __riscv_vmsne_vx_i8m1_b8(vand_m, 0, 32) instead. 
Thank you so much!

closed 2023-09-18 by Tameem97