riscv-non-isa/riscv-rvv-intrinsic-doc #286: vcreate intrinsics for LMUL > 1

zhongjuzhe, 2023-10-24:
We have vcreate/vget/vset for tuple type and vget/vset for LMUL > 1 type.

Why don't we have vcreate for LMUL > 1 ? 

vfloat16m8 =  __riscv_vcreate_v_f16m8 (vfloat16m1, ...);

eopXD, 2023-10-24:
Tuples are frequently created because users need to construct them as inputs for segment stores, and we have them because users do raise an issue that this will help them.

On the other hand, are users motivated to construct values of a higher LMUL upon their results? Should they be considering to use a higher LMUL at the first place?

eopXD, 2023-10-24:
`vget`/`vset` are there for functional completeness. `vcreate` is a syntax sugar though.

zhongjuzhe, 2023-10-24:
So, why don't LMUL > 1 has sugar too ?

eopXD, 2023-10-24:
I guess I don't have a strong reason against this, but I think it would be good to back this with motivation.

eopXD, 2023-10-25:
I guess I found a motivation...
https://github.com/opencv/opencv/blob/master/modules/core/include/opencv2/core/hal/intrin_rvv_scalable.hpp#L497

zhongjuzhe, 2023-10-25:
> I guess I found a motivation... https://github.com/opencv/opencv/blob/master/modules/core/include/opencv2/core/hal/intrin_rvv_scalable.hpp#L497

Yes. That's why I said we need vcreate for big LMUL.
I just remember long time ago some body ask me whether we can have vcreate like ARM SVE for big LMUL and tuple.

I just noticed you only add tuple vcreate recently.

nick-knight, 2023-10-25:
I don't agree with the hypothesis of this issue.

I agree that proposed `vcreate` for the (_non-tuple_) LMUL > 1 case would be syntactic sugar. 

However, `vcreate` is _not_ syntactic sugar for tuple types: rather, it is functionally necessary. To verify this yourself, try to initialize a tuple via a sequence of `vset` intrinsics, and compile with `-Wall -Werror`.

The root cause of this discrepancy is that the current API provides `vundefined` objects for non-tuple types, but not for tuple types. (I don't know whether or not this omission was intentional.)

Another approach towards consistency would be to add the "missing" `vundefined` tuple objects, and remove `vcreate`.

PR #288 (2023-10-30): Add `vcreate` for non-tuple types and `vundefine` for tuple types
Although this use case [0] is SIMD style and not that ideal for RVV, they can benefit from this syntax sugar.
On the other hand, `vundefined` for tuple types will help declaration under `-Winitialized` to avoid using a `vcreate` with numbers of `vundefined` non-tuple members in it.

Resolves #286.

For the `vcreate` addition, variants of an assembling a fractional LMUL type with smaller fractional LMUL, and assembling LMUL>1 type with fractional LMUL is omitted because they are unlikely use cases.

LLVM implementations of the ad [... cut]
closed 2023-10-30 by eopXD