riscv-non-isa/riscv-rvv-intrinsic-doc #299: Constraint of vector types in Zve32*

jacquesguan, 2023-11-21:
In this doc: https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/main/doc/rvv-intrinsic-spec.adoc
There is a constraint:
Types with subscript "*" is available when ELEN >= 64 (that is, unavailable under Zve32* and require at least Zve64x).
[image]

Why these vector types are not available for Zve32*?

topperc, 2023-11-21:
This sentence from section 3.4.2 in the spec removes LMUL=1/8 for ELEN=32. "For standard vector extensions with ELEN=32, fractional LMULs of 1/2 and 1/4 must be supported. For standard vector extensions with ELEN=64, fractional LMULs of 1/2, 1/4, and 1/8 must be supported"

The other text around there explains the rest.

The summary is that there no fractional LMULs with SEW==ELEN. And the fractional types that exist are the ones needed to be able to widen to LMUL==1 SEW==ELEN.

jacquesguan, 2023-11-21:
> This sentence from section 3.4.2 in the spec removes LMUL=1/8 for ELEN=32. "For standard vector extensions with ELEN=32, fractional LMULs of 1/2 and 1/4 must be supported. For standard vector extensions with ELEN=64, fractional LMULs of 1/2, 1/4, and 1/8 must be supported"
> 
> The other text around there explains the rest.
> 
> The summary is that there are no fractional LMULs with SEW==ELEN. And the fractional types that exist are the ones needed to be able to widen to LMUL==1 SEW==ELEN.

Thanks to response. So `vfloat16mf4_t ` and `vfloat32mf2_t` should also be not available for Zve32f, Zve32d, but for Zve64f, are they available too?

rofirrim, 2024-11-18:
Hi @jacquesguan,

> Thanks to response. So vfloat16mf4_t and vfloat32mf2_t should also be not available for Zve32f, but for Zve64f, are they available too?

Indeed this is what the spec says now.

[image]

Let me know this solves your question.

jacquesguan, 2024-11-19:
> Zve32f

Thank you, my question resolved.

closed 2024-11-19 by jacquesguan