riscv-non-isa/riscv-rvv-intrinsic-doc #310: Does `__riscv_v_intrinsic >= 1000000` imply overloaded intrinsics are supported?

camel-cdr, 2024-01-10:
Does `__riscv_v_intrinsic >= 1000000` imply overloaded intrinsics are supported?

There used to be a mention of `__riscv_v_intrinsic_overloading` in the rfc, but it doesn't seem to be in the spec anymore.
Can we rely on `__riscv_v_intrinsic >= 1000000` when using overloaded intrinsics?

rofirrim, 2024-11-18:
Hi @camel-cdr,

I think this assumption is safe to make.

Maybe `clang` should not set (or code should not rely on) the `__riscv_v_intrinsic_overloading` macro anymore unless there is some compatibility concern? /cc @4vtomat @topperc @kito-cheng

4vtomat, 2024-11-18:
I think there's no compatibility issue and we can bump the version to v1.0 at the same time~

4vtomat, 2024-11-22:
Resolved by https://github.com/llvm/llvm-project/pull/116597

closed 2024-11-22 by 4vtomat