riscv-non-isa/riscv-rvv-intrinsic-doc #318: Clarify the consequences of vxsat not being handled by the intrinsics

rofirrim, 2024-04-29:
As pointed by the ARC review it is acceptable that we leave out the vxsat support for a 1.0 version of the specification but we should clarify the consequences. The spec already contains a reference to this but it seems reasonable to expand on the consequences.

In summary: The specification does not provide a mechanism to read `vxsat` or update it in a predictable way using a sequence of fixed point arithmetic intrinsics.

rofirrim, 2024-05-03:
Earl from ARC suggests that

> Might I suggest that you specify for the intrinsics that would affect vxsat, that vxsat is UNSPECIFIED after execution of the intrinsic (to allow forward compatibility with whatever may or may not be done on the future to more properly support vxsat)?

closed 2024-05-08 by rofirrim

dzaima, 2024-05-08:
As a side-note, I wonder if it even makes sense for intrinsics to state anything about the `vxsat` CSR - clang [already](https://godbolt.org/z/93x7xs91K) can autovectorize plain scalar code to `vxsat`-modifying code, so it's already the case that basic regular C arithmetic can modify `vxsat`, thus making intrinsics also doing such moot.

nick-knight, 2024-05-08:
@dzaima I agree with a more general comment, that this API spec doesn't need to discuss any of the V-extension architectural state, in the same way that the C language specification of `<fenv>` doesn't mention `fcsr` and `frm`. Said differently, you could write a C compiler that retargets RVV intrinsics at a non-RVV ISA, emulating everything.

This type of discussion might be relevant in an ABI document, or a document about mixing RVV intrinsics and inline asm, but the RVV intrinsics spec (intentionally) avoids both of these topics. 

In the case of this PR, we are responding to a request from the RVI Architecture Review Committee that is blocking ratification of this spec. Rather than philosophizing and bickering with the ARC about what truly belongs in a C API spec, we decided it was more expedient to just simply implement their request, which amounts to expanding a (non-normative!) note that existed already.

PR #330 (2024-05-08): Clarify the current support of `vxsat`
Fixes #318