riscv-non-isa/riscv-rvv-intrinsic-doc #380: How to perform bit shift operations on mask type (such as vbool16_t) through intrinsic functions？

cocacola-wing, 2024-11-11:
Hi, there!

I used a mask type` vbool16_t` to represent the carry bit of addition, and now I need to shift it left to propagate the carry. But I can't find the corresponding intrinsic function, not even the intrinsic function to convert `vbool16_t` to `uint16_t`. 

I saw someone mentioned it, but it was in assembly code. 
[Rotate and shift vector mask bits](https://github.com/riscvarchive/riscv-v-spec/issues/919)

Is it necessary to add some pseudo intrinsics to achieve it?

topperc, 2024-11-11:
You can use `__riscv_vreinterpret_v_b16_u64m1` to cast from `vbool16_t` to `vuint64m1_t` then you can use __riscv_vsrl.

cocacola-wing, 2024-11-12:
> You can use `__riscv_vreinterpret_v_b16_u64m1` to cast from `vbool16_t` to `vuint64m1_t` then you can use __riscv_vsrl.

Oh! I didn't notice the super `vreinterpret`. Thank you so much!

cocacola-wing, 2024-11-12:
It works! Thanks again!

closed 2024-11-12 by cocacola-wing