riscv-non-isa/riscv-rvv-intrinsic-doc #381: Qestion about vfmv.v.f for vector bfloat16

ff520git, 2024-11-18:
There is a question about wheter vector bfloat16 extension supports the vfmv.v.f. 
In the "https://raw.githubusercontent.com/riscv-non-isa/rvv-intrinsic-doc/refs/heads/main/auto-generated/intrinsic_funcs.adoc", the intrinsic of vfmv.v.f supports float16 type. From my understanding,this instrction olny cares about the width of the data and does not need to care about the data type. Therefore,it should aslo support __bfloat16, but I didn't found it in the intrinsic_funcs.adoc of vector bfloat16.
So may I ask if it was missied or intentionally done? Thanks.

topperc, 2024-11-18:
If the 16-bit value in the FP register is not properly nan-boxed to FLEN bits, the vmv.v.f instruction would splat an fp16 nan instead of a bf16 nan. So for the hardware implementation, the data type does matter.

I'm not sure if its possible for the compiler to generate code that would create an improperly nan-boxed value.

ff520git, 2024-11-18:
Thanks for your answer. "the vfmv.v.f instruction would splat an fp16 nan instead of a bf16 nan." Is it specified in the spec? Is there a problem generating a canonical NaN value of bfloat16 here?Thanks.

topperc, 2024-11-18:
> Thanks for your answer. "the vfmv.v.f instruction would splat an fp16 nan instead of a bf16 nan." Is it specified in the spec? Is there a problem generating a canonical NaN value of bfloat16 here?Thanks.

From the vector spec "For floating-point operations, the scalar can be taken from a scalar f register. If FLEN > SEW, the value in the f registers is checked for a valid NaN-boxed value, in which case the least-significant SEW bits of the f register are used, else the canonical NaN value is used."

This doesn't explicitly say the fp16 canonical NaN is used, but it's the only thing that makes sense for the arithmetic instructions.

ff520git, 2024-11-18:
"but it's the only thing that makes sense for the arithmetic instructions." So can I understand is as for the mov operation, it doesn't care the data value in the scalar f register?

topperc, 2024-11-18:
> "but it's the only thing that makes sense for the arithmetic instructions." So can I understand it as for the mov operation, it doesn't care the data value in the scalar f register?

No. All instructions including vfmv.v.f with a scalar operand will create an fp16 canonical NaN if the F register is not properly NaN-boxed.

ff520git, 2024-11-19:
OK. Got it! Thanks.

topperc, 2024-11-19:
There was a recent proposal for full bf16 support shared to sig-vector by Andrew Waterman https://github.com/aswaterman/riscv-misc/blob/34ac270c6f227d5ed7008810ef8e703e1d06f408/isa/zvfbfa.adoc

The canonical NaN issue is mentioned in the footnotes.