riscv-non-isa/riscv-rvv-intrinsic-doc #385: Generating 'm1' Code Instead of 'mf8' for RISC-V Intrinsic in Assembly

aaaaala, 2025-01-19:
I use the intrinsic `__riscv_vmsgt_vv_i64m1_b64(vs2, vs1, vl)`, but the generated assembly code is `vsetvli a5, zero, e8, mf8, tu, mu`. What can I do to make the assembler generate code with 'm1' as the vector length parameter instead?
Complete code:
size_t vl = __riscv_vsetvl_e64m1(20);
vint64m1_t vs2 = __riscv_vundefined_i64m1();
vint64m1_t vs1 = __riscv_vundefined_i64m1();
vbool64_t tmp = __riscv_vmsgt_vv_i64m1_b64(vs2, vs1, vl);

dzaima, 2025-01-19:
With `riscv64-linux-gnu-gcc-14` (and -13) I get:

```asm
  vsetivli  a4,20,e8,mf8,ta,ma
  [...a bunch of scalar arith]
  vsetvli a5,zero,e8,mf8,tu,mu
  [...more scalar arith]
  vl1re64.v v2,0(a4)
  vl1re64.v v1,0(a5)
  [...and more scalar arith]
  vsetvli zero,a5,e64,m1,ta,ma
  vmsgt.vv  v1,v2,v1
  vsetvli a5,zero,e8,mf8,tu,mu
  vsetvli a5,zero,e8,mf8,ta,ma
```

So there still is a `vmsgt.vv` after a `vsetvl ...,e64,m1,...`, it's just some ways down. Without optimizations gcc just inserts additional instructions (and, importantly, `__riscv_vsetvl_e64m1(x)` returns the same value as `__riscv_vsetvl_e8mf8(x)` so gcc has decided to replace the former with the latter as on `-O0` it doesn't try to reuse `vsetvl`s).

aaaaala, 2025-01-20:
How can I make GCC only use` __riscv_vsetvl_e64m1(x)` ? Is there a way to achieve thi?

dzaima, 2025-01-20:
I wouldn't think so; compilers are free to do anything they want as long as the output is semantically equivalent. Same as that there isn't a way to force a compiler to run `int foo(int x) { return x * 1; }` via a `mul` instruction.

aaaaala, 2025-01-21:
You're right. I may have misunderstood the requirements of the assignment earlier. Thank you for your answer!

closed 2025-01-21 by aaaaala