riscv-non-isa/riscv-rvv-intrinsic-doc #387: vl parameter in vmv.s.x?

zingaburga, 2025-02-16:
The spec says that `vmv.s.x` does not have a `vl` parameter:

https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/bfffe11a6960aab40951c9496631771e83c97cae/doc/rvv-intrinsic-spec.adoc?plain=1#L72

However, all the function prototypes for `__riscv_vmv_s_x_*` list `vl` as a parameter.

https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/bfffe11a6960aab40951c9496631771e83c97cae/auto-generated/intrinsic_funcs/07_vector_permutation_intrinsics.adoc?plain=1#L40-L52

---

Also the `__riscv_vmv_x_s_*` declarations [shown here](https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/bfffe11a6960aab40951c9496631771e83c97cae/doc/rvv-intrinsic-spec.adoc#scalar-move-instructions) do not match what's in the main listing.

dzaima, 2025-02-16:
The `vl` parameter for `__riscv_vmv_s_x_*` is necessary for the case of `vl=0`, for which the instruction gives an entirely arbitrary result (or in the case of the `_tu` version, returns `vd` unmodified); very annoying/weird, but that's unfortunately what the ratified RVV spec specifies for reasons I don't know.

(the `__riscv_vmv_x_s_i8m1_i8` & co declaration examples in `rvv-intrinsic-spec.adoc` do seem wrong though)

topperc, 2025-02-17:
I suspect the note was supposed to be vmv.x.s not vmv.s.x

zingaburga, 2025-02-17:
Thanks for the fixes!

closed 2025-02-17 by zingaburga