riscv-non-isa/riscv-rvv-intrinsic-doc #397: Discussion about zvfbfa intrinsics

4vtomat, 2025-05-13:
When following the naming rule, some of instructions in Zvfbfa will be ambiguous with float16 version. For example intrinsic for `vfncvt.x.f.w` float16 version would be
```
vint8mf8 __riscv_vfncvt_x_f_w_i8mf8(vfloat16mf4 src, size_t vl);
```
however intrinsic for bfloat16 version would be
```
vint8mf8 __riscv_vfncvt_x_f_w_i8mf8(vbfloat16mf4 src, size_t vl);
```
which is exactly the same thus ambiguous.
After discussion offline with colleague, we came up with 1 thought: Add `_alt` suffix to bfloat16 version intrinsic, i.e. `__riscv_vfncvt_x_f_w_i8mf8_alt`.
However we have several instructions that encounter this issue, so not sure if it's better to add `_alt` suffix to all of bfloat16 intrinsics or just add the suffix to those intrinsics that are ambiguous?

cc: @topperc @kito-cheng @rofirrim

rofirrim, 2025-05-13:
If it serves as an existing experience data point, when we implemented this (it was part of a project) in our old compiler for RVV0.7.1 we also faced this issue and we ended adding the explicit type.

For instance in widenings we did this

```
__epi_4xf32 __builtin_epi_vfwadd_4xf32(__epi_4xf16 a, __epi_4xf16 b,
                                       unsigned long int gvl);
__epi_4xf32 __builtin_epi_vfwadd_4xf32_bf16(__epi_4xbf16 a, __epi_4xbf16 b,
                                            unsigned long int gvl);
```

and in `vfncvt.f.f` (looks like we ddin't implement `vfncvt.x.f`, we probably deprioritized for this PoC) looked like this

```
__epi_8xf8_143 __builtin_epi_vfncvt_f_f_8xf8_143_8xf16(__epi_8xf16 a,
                                                       unsigned long int gvl);
__epi_8xf8_152 __builtin_epi_vfncvt_f_f_8xf8_152_8xbf16(__epi_8xbf16 a,
                                                        unsigned long int gvl);
```
You can take a look here https://admin.hca.bsc.es/epi/ftp/doc/intrinsics/EPI-0.7/epi-intrinsics.html

(please mind that we used a very different convention to the one actually used by the RVV intrinsics)

4vtomat, 2025-06-02:
> Apologies for the delay.
> 
> > One question on your approach: how does the masking version look like in your example?
> 
> it looked like this (because 0.7.1 was conceptually only `ta,mu`)
> 
> __epi_16xf8_143
> __builtin_epi_vfncvt_f_f_16xf8_143_16xf16_mask(__epi_16xf8_143 merge,
>                                                __epi_16xf16 a, __epi_16xi1 mask,
>                                                unsigned long int gvl);
> In the RVV intrinsics I presume the suffix (`_m`, `_tu`, etc) could go after the second type?

I see, but does it break the rules of naming scheme in https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/main/doc/rvv-intrinsic-spec.adoc#explicit-non-overloaded-naming-scheme? Since the only type suffix should be return type, however in your approach, it needs both "return type" and "input type" as suffix lol

4vtomat, 2025-06-12:
> 
> and in `vfncvt.f.f` (looks like we ddin't implement `vfncvt.x.f`, we probably deprioritized it for this PoC) looked like this
> 
> ```
> __epi_8xf8_143 __builtin_epi_vfncvt_f_f_8xf8_143_8xf16(__epi_8xf16 a,
>                                                        unsigned long int gvl);
> __epi_8xf8_152 __builtin_epi_vfncvt_f_f_8xf8_152_8xbf16(__epi_8xbf16 a,
>                                                         unsigned long int gvl);
> ```
> 

@rofirrim Hey Roger, seems in your document it's only non-overloaded version, I'd like to know what does overloaded version of this intrinsics look like in your approach?

rofirrim, 2025-06-16:
> [@rofirrim](https://github.com/rofirrim) Hey Roger, seems in your document it's only non-overloaded version, I'd like to know what does overloaded version of this intrinsics look like in your approach?

Hi @4vtomat, we never implemented overloaded versions for that PoC, it was always explicit naming for us. Sorry.