riscv-non-isa/riscv-rvv-intrinsic-doc #402: Suspected Incorrect Results from __riscv_vfmul_vv_f32m4 for Tuples Beyond the First

soniab, 2025-06-27:
Hello, 

I am using the following test snippet:

#include <stdio.h>
#include <stdlib.h>
#include <riscv_vector.h>

int main() {
    const int D = 8;
    size_t vl = __riscv_vsetvl_e32m1(D);    // VL for each subvector
    size_t vl_m4 = __riscv_vsetvl_e32m4(D); // VL for m4

    // Create 4 subvectors for Q
    float q0[D], q1[D], q2[D], q3[D];
    for (int i = 0; i < D; i++) {
        q0[i] = 5.0f;
        q1[i] = 2.0f;
        q2[i] = 3.0f;
        q3[i] = 4.0f;
    }

    // Create 4 subvectors for K
    float k0[D], k1[D], k2[D], k3[D];
    for (int i = 0; i < D; i++) {
        k0[i] = 10.0f;
        k1[i] = 20.0f;
        k2[i] = 30.0f;
        k3[i] = 40.0f;
    }

    // Load Q subvectors
    vfloat32m1_t vq0 = __riscv_vle32_v_f32m1(q0, vl);
    vfloat32m1_t vq1 = __riscv_vle32_v_f32m1(q1, vl);
    vfloat32m1_t vq2 = __riscv_vle32_v_f32m1(q2, vl);
    vfloat32m1_t vq3 = __riscv_vle32_v_f32m1(q3, vl);

    // Load K subvectors
    vfloat32m1_t vk0 = __riscv_vle32_v_f32m1(k0, vl);
    vfloat32m1_t vk1 = __riscv_vle32_v_f32m1(k1, vl);
    vfloat32m1_t vk2 = __riscv_vle32_v_f32m1(k2, vl);
    vfloat32m1_t vk3 = __riscv_vle32_v_f32m1(k3, vl);

    // Pack into m4 vectors (this is like creating "tuples")
    vfloat32m4_t vQ = __riscv_vundefined_f32m4();
    vQ = __riscv_vcreate_v_f32m1_f32m4(vq0,vq1,vq2,vq3);

    vfloat32m4_t vK = __riscv_vundefined_f32m4();
    vK = __riscv_vcreate_v_f32m1_f32m4(vk0, vk1,vk2,vk3);

    // Multiply all tuples
    vfloat32m4_t vResult = __riscv_vfmul_vv_f32m4(vQ, vK, vl_m4);

    // Extract and print each tuple
        vfloat32m1_t vrow = __riscv_vget_v_f32m4_f32m1(vResult, 1);
        float res[D];
        __riscv_vse32_v_f32m1(res, vrow, vl);
        for (int i = 0; i < D; i++) {
            printf("%6.1f ", res[i]);
        }
        printf("\n");

    return 0;
}

I created a minimal example to test the behavior of the __riscv_vfmul_vv_f32m4 intrinsic. The motivation was to validate its correctness after observing unexpected results in my benchmarks. The test confirmed that I am indeed getting incorrect outputs from __riscv_vfmul_vv_f32m4

Specifically, the result for the first tuple (index 0) is correct: for example: vfloat32m1_t vrow = __riscv_vget_v_f32m4_f32m1(vResult, 0); produces expected values. However, subsequent tuples (e.g., indices 1, 2, and 3) yield incorrect results, consistently across multiple runs. 

I have tested this with different compilers, and the issue persists.

**Compilation command:**
clang --target=riscv64-unknown-linux-gnu -march=rv64gcv_zvfh  -static   test.c

**clang version:**
clang version 20.0.0git (https://github.com/llvm/llvm-project.git e2dc50c92987e6886b1b1641ebdbe9cc53bae192)
Target: /cfs/klemming/projects/snic/naiss2023-23-linux-gnu-49/sonia/llvm-RISCV/RISCV/riscv64-unknown
Thread model: posix

**With GCC:**
riscv64-unknown-linux-gnu-gcc -march=rv64gcv_zvfh  -static   test.c

**GCC version:**
riscv64-unknown-linux-gnu-gcc (g04696df0963) 14.2.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Command to run:
qemu-riscv64 -cpu rv64,v=true,vext_spec=v1.0,zfh=true,vlen=1024,elen=64 ./a.out

dzaima, 2025-06-27:
Yeah, that'll behave exactly the same.

The `vset_v`/`vcreate_v`/`vget_v` intrinsics operate over `VLEN*LMUL`-sized vectors/vector parts always, they have no way to know that you want to pack exactly at 256-bit boundaries; they compile to only just rearranging registers, which will by necessity always be `VLEN`-sized.

If you want to utilize `vlen=1024`, a single `m1` vector running on vlen=1024 hardware/emulator will already give you a 1024-bit vector, no need to use `m4`. Alternatively, you can use `m4` on `vlen=256`, or even `m8` on `vlen=128`.

If you want to merge multiple vectors exactly on 256-bit boundaries in a VLEN-agnostic way, you'll by necessity need to use `vslideup` & blending & whatnot.

soniab, 2025-06-27:
Thank you for the clarification.